SimplifiedLogic/creoson #79: Param List Error on Creo version 9

btxp, 2022-05-25:
I have been getting the following error when using the "parameter", "List" Function only Creo 9 is affected (version 3 & 8 are unaffected) I have tried the using the new feature "creo", "set_creo_version" but still get the same error

{"status":{"error":true,"message":"A PTC error has occurred when retrieving parameter DESCRIPTION: Obsolete Functionality.  Check your model/drawing for accuracy."},"data":null}

Anybody help with this?
Thanks in advance

adama2000, 2022-05-26:
We haven't yet tested creoson with Creo 9, some of the functionality may
have changed.  I'll look into it.

adama2000, 2022-08-26:
In Creo 9 the standard JLink call (Parameter.GetValue()) we use to get a
parameter value now returns that "Obsolete Functionality" error.  There is
an alternate function Parameter.GetScaledValue() which still works, but
we're going to have to convert the code to use it.

Adam Andrews

adama2000, 2023-02-15:
Just posted a new release of Creoson which works with Creo 9.  The Obsolete
Functionality error will only occur if your PATH is still pointing to an
older version of Creo, so make sure you change the path using the
CreosonSetup app, or by editing the setvars.bat file.

Adam Andrews