SimplifiedLogic/creoson #80: Creoson server crashes

stekicar, 2022-06-02:
I am going through large numbers of components, adding material and parameters (100+) and Creoson server crashes:

`Traceback (most recent call last):
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\urllib3\connectionpool.py", line 699, in urlopen
    httplib_response = self._make_request(
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\urllib3\connectionpool.py", line 445, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\urllib3\connectionpool.py", line 440, in _make_request
    httplib_response = conn.getresponse()
  File "C:\Program Files\Python310\lib\http\client.py", line 1368, in getresponse
    response.begin()
  File "C:\Program Files\Python310\lib\http\client.py", line 317, in begin
    version, status, reason = self._read_status()
  File "C:\Program Files\Python310\lib\http\client.py", line 278, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "C:\Program Files\Python310\lib\socket.py", line 705, in readinto
    return self._sock.recv_into(b)
ConnectionResetError: [WinError 10054] An existing connection was forcibly closed by the remote host

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\requests\adapters.py", line 439, in send
    resp = conn.urlopen(
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\urllib3\connectionpool.py", line 755, in urlopen
    retries = retries.increment(
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\urllib3\util\retry.py", line 532, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\urllib3\packages\six.py", line 769, in reraise
    raise value.with_traceback(tb)
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\urllib3\connectionpool.py", line 699, in urlopen
    httplib_response = self._make_request(
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\urllib3\connectionpool.py", line 445, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\urllib3\connectionpool.py", line 440, in _make_request
    httplib_response = conn.getresponse()
  File "C:\Program Files\Python310\lib\http\client.py", line 1368, in getresponse
    response.begin()
  File "C:\Program Files\Python310\lib\http\client.py", line 317, in begin
    version, status, reason = self._read_status()
  File "C:\Program Files\Python310\lib\http\client.py", line 278, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "C:\Program Files\Python310\lib\socket.py", line 705, in readinto
    return self._sock.recv_into(b)
urllib3.exceptions.ProtocolError: ('Connection aborted.', ConnectionResetError(10054, 'An existing connection was forcibly closed by the remote host', None, 10054, None))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\creopyson\connection.py", line 53, in _creoson_post
    r = requests.post(self.server, data=json.dumps(request))
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\requests\api.py", line 117, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\requests\api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\requests\sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\requests\sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\requests\adapters.py", line 498, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', ConnectionResetError(10054, 'An existing connection was forcibly closed by the remote host', None, 10054, None))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\python_code\change_data_to_models.py", line 154, in <module>
    creopyson.file.open_(srv, comp, dirname=None, generic=None, display=True, activate=True, new_window=True, regen_force=None)
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\creopyson\file.py", line 797, in open_
    return client._creoson_post("file", "open", data)
  File "C:\Users\user\AppData\Roaming\Python\Python310\site-packages\creopyson\connection.py", line 55, in _creoson_post
    raise ConnectionError(e)
ConnectionError: ('Connection aborted.', ConnectionResetError(10054, 'An existing connection was forcibly closed by the remote host', None, 10054, None))`

I tried to slow down loop by adding 2 seconds but it did not help.

I guess, I will have to check if connection is present at each loop and if not start creoson server within the code.

stekicar, 2022-06-02:
How can I start creoson server from python code?

Zepmanbc, 2022-06-02:
this may help: https://github.com/Zepmanbc/creopyson/issues/14#issuecomment-704730644