SimplifiedLogic/creoson #85: Creoson application suspended / not responding

girishdubey, 2022-10-14:
Good day

Applications Creo and Creoson begin in a subprocess. Once creoson has started, I just check to see if creo is running. Initially, I receive the following error: "Unable to connect to Creo over J-Link." True After 3 seconds of retrying, I was suspended/not responding from creoson, and I was unable to access the creo web service.

Working in separate processes is effective. Can somebody explain what's happening?

Note: Creo is taking time to open (~5-10 seconds)

kind regards

DieSwartKat, 2022-10-14:
Hi
I have found that it's important that CREO run before trying to start CREOSON. If I force a connection to CREO before it properly start then CREO crash. (using CREO 3 m190)
I should also point out that I've tried using vscode for my testing and it seems like VSCODE control calls in a separate thread which if stopped stop everything it started - might be worth considering if the way to run these subprocess could close CREOSON.
If I recall correctly there is a way to make subprocess start an application by using pop ? Might have to check the documentation.

Zepmanbc, 2022-10-14:
I confirm what @DieSwartKat said, Creo must be running before you start Creoson
Are you sure there is only one Creo running?
I use VSCode most of the time to run scripts but never had any trouble because of VSCode

DieSwartKat, 2022-10-14:
I have however been able to restart CREO while CREOSON was running and it connected without any issues. I'm wondering if it's not related to the license server in the background? First startup has the server running and then if I restart CREO the license server is up and running already and connection can work. (I'm speculating)

DieSwartKat, 2022-10-17:
I've had success starting CREOSON before CREO, but only once I've start CREO once before and existed it.  I think it might be related to the licensing. As CREO typically starts faster the second time round.
VSCODE seems to do some fancy threading when using debug and if you start CREO using your your code in debug model, it will kill CREO once the code has completed.
Best advice is to start CREO outside of VSCODE. And to start CREO before starting CREOSON. Although there is exceptions to this rule which is not obvious.