SimplifiedLogic/creoson #93: regen_failure_handling is not reset

broooooot, 2023-03-29:
Hi there,

we have the problem here that the parameter `regen_failure_handling` is set to `resolve_mode`, apparently by creoson. When I googled it I stumbled upon the source code of `JlinkUtils.java` where this config option is mentioned. Unfortunately, we can not work with this option enabled and sometimes it does not set itself back to `no_resolve_mode`. 

I can not say when this happens exactly and unfortunately, I don't really have time to do that here at work, but maybe, you find the error pretty fast. 

Thank you very much! If I can help in any way I will of course try.

adama2000, 2023-04-03:
The following creoson functions update the regen_failure_handling option:
- familytable.replace
- feature.delete
- feature.resume
- feature.suppress
- file.regenerate
- file.assemble (when suppress=true)

I suppose one answer might be that after running all other creoson
functions, your code makes a creoson call to "creo.set_config" to set that
option deliberately.  You MIGHT get a "deprecated option" error when you do
that if previous creoson calls haven't set that option, because creoson
disables that error when it sets the option internally.

Do you get any errors from creoson?

I'll look into it further...

Adam Andrews

broooooot, 2023-04-03:
Hi Adam,

thank you for your response! Strangely though I'm not using any of these functions in the code but only the following:
```
creo_set_creo_version()
creo_pwd()
file_get_active()
creo_cd()
creo_list_files()
file_open()
interface_mapkey()
interface_export_image()
interface_export_pdf()
interface_export_file()
parameter_exists()
drawing_select_sheet()
drawing_create()
view.list_()
view.activate()
drawing_get_cur_model()
drawing_get_num_sheets()
file_close_window()
parameter.list_()
```
As a workaround I already tried setting the config option via creoson but I'm getting a runtime error saying `Please check your config option value: no_resolve_mode` so I'm guessing, because it's deprecated, you can not set it via creoson. But, as a quick fix, I set the parameter in the config.sup file so it can (presumably) not be overwritten and since then we haven't had any occurence of the parameter being set back to `resolve_mode` and the scripts run fine anyways.

I hope that helps, thank you!

Zepmanbc, 2023-04-03:
Hi,
I do something like this when I instanciate the client at the begining of my script
```self.client.creo_set_config("regen_failure_handling", "no_resolve_mode")```
it's ok with Creo 8, did not tried with Creo 9

jordifas, 2023-06-22:
I have the same issue. Where can I find the config.sup file?

eng-marani, 2023-11-22:
Unfortunately, your product will likely no more work for Creo10 upward.
Ptc has simply removed the regen_failure_handling option alltogether.
JLink/VB API now will have all functions that relies on the resolve mode,
no longer working, but throwing an exception XToolkitBadContext instead.

Apparently, there is nothing we can do to restore functionality in Creo10.
If you can find any workaround please let me know. So sorry about this.

davidhbigelow, 2023-11-25:
Creo 10 support is coming!  We have been busy making sure other products were supported first.

eng-marani, 2023-11-27:
How did you overcome the recent limitation of all .ExecuteFeatureOps() that return XToolkitBadContext
when performing various feature operations in Creo10, like .CreateDeleteOp() for removing a component ?
I had to resort to a mapkey (selection+ProCmdEditDelete) to workaround this.

adama2000, 2023-12-11:
New creoson release was posted this weekend.  If you set_creo_version with
9 or 10, it will now use an embedded native DLL to do certain feature
operations, and will no longer try to do the JLink FeatureOps calls you
mentioned.

Adam