SimplifiedLogic/creoson #94: “JLTransform”，How do you use it?

HGQ21102, 2023-04-06:
“JLTransform”，How do you use it?

adama2000, 2023-04-07:
Are you looking for a description, or for examples?  I think there are a
couple of examples in the creoson documentation that comes with the
software.

The JLTransform class represents the PTC J-Link "Transform3D" class; it
contains an origin and x/y/z vectors, which are gotten from the Transform3D
object; it also contains x/y/z rotations which are computed by creoson.

JLTransform is passed (optionally) as INPUT to the file.assemble function.
It controls the component's orientation/position after assembly.

JLTransform is returned (optionally) as OUTPUT from the bom.get_paths()
function.  It's only returned if you set get_transforms to true on the
call.  It returns the transformation matrix from the assembly's origin to
the component.

Adam Andrews

HGQ21102, 2023-04-07:
Sorry, my English is not good. Can you give me an example？Current universal translation software

0781532, 2023-06-13:
I got the same problem when I want to transform a component after assembling to an .asm.

I tried to write code to conduct this work but not succeed.

Could you please deliver a example for this function (file_assemble or file_get_transform)

Thank you!
L

What I did:

------------------------------
request_data = {
        "sessionId": "~sessionId~",
        "command": "file",
        "function": "assemble",
        "data": {
            "file": "adl_1744_50x25_4a.prt",
            "into_asm": "a2.asm",
            "transform": [ 54, 23, 45 ],                      
            
        }
    }

out1 = c.file_assemble("adl_1744_50x25_4a.prt", into_asm="A2.asm", transform = list(request_data['data']['transform']) )
------------------------------
and

---------------------------------
{'featlist': [{'name': 'adl_1744_50x25_4a.prt', 'type': 'csys', 'feat_id': 4, 'path': [{54, 23, 45}]}]}
    f=c.feature_user_select_csys() 
    print (c.file_get_transform(csys=f['featlist'][0]))
---------------------------------

RuntimeError: Invalid input data: [54, 23, 45]