SimplifiedLogic/creoson #99: deal with multiple instances

tiptoptom, 2023-08-31:
I often work with several Creo sessions at once. Unfortunately, in this case I always get the following error from Creoson: 
`RuntimeError: Unable to connect to Creo; more than one instance of Creo is running`

Is it possible to use Creoson with multiple Creo sessions or would that be feasible?

eng-marani, 2023-09-05:
I think this is a limitation of the "async" application model that J-Link uses when Creoson self-hosting server connects to Creo. Only a single instance is supported for the "async" programming model, that is the only available in J-Link (now renamed as "java free" in Creo 10+). I suspect no workaround may exist there, unfortunately.

tiptoptom, 2023-09-05:
In that case, do you see a way to drive a single session with Creoson when multiple sessions are open?

eng-marani, 2023-09-06:
I have no idea on how to steer the request to the proper creo instance. Also, identify the proper instance would be problematic, because the only discriminant would be the process id. In my humble opinion, support of multiple instances for the async application model require PTC engineers to extend or fix the creo codebase, and this is unlikely in a short timeframe. 

With old good COM technology (proprietary Microsoft) i shall use the ROT-Table, using wich you can enumerate all running COM server instances and connect to a specific one, but i think nothing similar actually exist for J-link connections in creo. 

As far as i know, the Creo instance do not provide direct API access, but relies on a singleton instance of a different running executable, pro_comm_msg.exe, that supposedly must be aware of connecting with multiple instance, but we do not have any control over it.

So, to have this feature implemented, must file a request directly to ptc support, i think.

closed 2026-08-20 by davidhbigelow