SimplifiedLogic/creoson #104: Unable to connect to Creo through J-Link

slinkytoad, 2024-01-15:
I had this working last week, and now this week I am running into issues. As far as I know, nothing has changed in my setup.

I am working on using creoson with powershell, which shouldn't be an issue, as powershell can talk JSON with ease.

I am able to connect and receive a sessionId from creoson. When I go to run any other function I get an error:

`Unable to connect to Creo through J-Link`

When I run the creoson server via command line I get this error:
```
com.ptc.pfc.Implementation.pfcExceptions$XToolkitNotFound
        at com.ptc.pfc.pfcAsyncConnection.pfcAsyncConnection.AsyncConnection_Connect(Native Method)
        at com.simplifiedlogic.nitro.jlink.calls.asyncconnection.CallAsyncConnection.connect(CallAsyncConnection.java:47)
        at com.simplifiedlogic.nitro.util.JLConnectionUtil.createConnectionEntry(JLConnectionUtil.java:184)
        at com.simplifiedlogic.nitro.util.JLConnectionUtil.makeAsyncConnection(JLConnectionUtil.java:159)
        at com.simplifiedlogic.nitro.rpc.JLISession.setConnection(JLISession.java:172)
        at com.simplifiedlogic.nitro.rpc.JLISession.getConnectionId(JLISession.java:180)
        at com.simplifiedlogic.nitro.jlink.impl.JLView.list(JLView.java:72)
        at com.simplifiedlogic.nitro.jlink.impl.JLView.list(JLView.java:53)
        at com.simplifiedlogic.nitro.jshell.json.handler.JLJsonViewHandler.actionList(JLJsonViewHandler.java:71)
        at com.simplifiedlogic.nitro.jshell.json.handler.JLJsonViewHandler.handleFunction(JLJsonViewHandler.java:54)
        at com.simplifiedlogic.nitro.jshell.json.JShellJsonHandler.handleRequest(JShellJsonHandler.java:197)
        at com.simplifiedlogic.nitro.jshell.json.JShellJsonHandler.handleRequest(JShellJsonHandler.java:136)
        at com.simplifiedlogic.nitro.jshell.JshellHttpHandler.handle(JshellHttpHandler.java:70)
        at com.sun.net.httpserver.Filter$Chain.doFilter(Filter.java:77)
        at sun.net.httpserver.AuthFilter.doFilter(AuthFilter.java:83)
        at com.sun.net.httpserver.Filter$Chain.doFilter(Filter.java:80)
        at sun.net.httpserver.ServerImpl$Exchange$LinkHandler.handle(ServerImpl.java:677)
        at com.sun.net.httpserver.Filter$Chain.doFilter(Filter.java:77)
        at sun.net.httpserver.ServerImpl$Exchange.run(ServerImpl.java:649)
        at sun.net.httpserver.ServerImpl$DefaultExecutor.execute(ServerImpl.java:158)
        at sun.net.httpserver.ServerImpl$Dispatcher.handle(ServerImpl.java:433)
        at sun.net.httpserver.ServerImpl$Dispatcher.run(ServerImpl.java:398)
        at java.lang.Thread.run(Thread.java:745)
```

Any thoughts are helpful, we are trying to replace our use of the Jerand plugin for exporting to ProNest.

Here are my functions and code from powershell for posterity:
```
function connect-creoson {
    $var = @{
        command = "connection"
        function = "connect"
    }

    $json = ConvertTo-Json($var)

    $result = Invoke-RestMethod -uri "http://localhost:9056/creoson" -Method POST -Body $json

    $result.sessionId
    
}

function disconnect-creoson {
    param (
        $session
    )
    
    $var = @{
        sessionId = $session
        command = "connection"
        function = "disconnect"
    }

    $json = ConvertTo-Json($var)

    $result = Invoke-RestMethod -uri "http://localhost:9056/creoson" -Method POST -Body $json

    if ($result.status.error) {
        Write-Host "Error when disconnecting from Creoson"
    }else {
        Write-Host "Disconnected Successfully"
    } 
}

function get-simReps {
    param (
        [Parameter(Mandatory=$true)] $session
    )
    
    $var = @{
        sessionId = $session
        command = "file"
        function = "list_simp_reps"
    }

    $json = ConvertTo-Json($var)

    $result = Invoke-RestMethod -Uri "http://localhost:9056/creoson" -Method POST -Body $json

    $result
}

function get-creosonViews {
    
    $var = @{
        sessionId = $session
        command = "view"
        function = "list"
    }
    
    $json = ConvertTo-Json($var)

    $result = Invoke-RestMethod -Uri "http://localhost:9056/creoson" -Method POST -Body $json

    $result
}

# connect first

$session = connect-creoson

# grab what simprep the user wants to use

$simpreps = get-simReps

disconnect-creoson $session
```

davidhbigelow, 2024-01-15:
Seems like you have a service still running or J-LINK is not installed.

You might try a complete reboot of your machine.

If you upgraded Creo - you should dbl check that J-LINK is installed... and
that you are talking to the correct version of Creo (e.g. Creo 8 vs 10)

Dave

slinkytoad, 2024-01-15:
I did make sure J-LINK is installed, that was an issue when I first started with creoson. 

I did try a restart, as I saw another issue about windows sleep causing issues. 

I am going to try a reinstall of Creo and see what happens.

adama2000, 2024-01-17:
The "XToolkitNotFound" error usually means that Creo isn't running, that
J-Link couldn't find a current instance of Creo to connect to.

Adam

closed 2026-08-20 by davidhbigelow