SimplifiedLogic/creoson #108: creoson_run.bat

yuyaU0507, 2024-03-06:
Hi there,

I'd like to start the creoson server to automatically to build a fully automated program.
I found "creoson_run.bat" and "setvars.bat" in the creoson folder so tried to run them.
When I run "creoson_run.bat" and got a error below:

java.net.BindException: Address already in use: bind
        at sun.nio.ch.Net.bind0(Native Method)
        at sun.nio.ch.Net.bind(Net.java:444)
        at sun.nio.ch.Net.bind(Net.java:436)
        at sun.nio.ch.ServerSocketChannelImpl.bind(ServerSocketChannelImpl.java:214)
        at sun.nio.ch.ServerSocketAdaptor.bind(ServerSocketAdaptor.java:74)
        at sun.net.httpserver.ServerImpl.<init>(ServerImpl.java:100)
        at sun.net.httpserver.HttpServerImpl.<init>(HttpServerImpl.java:50)
        at sun.net.httpserver.DefaultHttpServerProvider.createHttpServer(DefaultHttpServerProvider.java:35)
        at com.sun.net.httpserver.HttpServer.create(HttpServer.java:129)
        at com.simplifiedlogic.nitro.jshell.MainServer.main(MainServer.java:102)

I couldn't find what is wrong with the bat file honestly.
May anyone help me?

Thanks!

nirav-98, 2024-03-06:
I'm not expert, but I'm also using creoson using script. 
I have observed that it needs a manual start for the first time, since it asks the location of Creo (C://path-to-creo-common-files). 
it works with the .bat file later on. 
Also, it looks like you have some port issue, can you try checking if that port is occupied, or killing that process or changing the port settings etc

adama2000, 2024-03-10:
This error means the port is already in use.  There are two possible reasons:

1) You already have an instance of the Creoson server running.  If so, you can just use that one.  If you want to kill the other Creoson server, you'll need either run the CreosonSetup GUI app and click the "Stop CREOSON" button, or use the Task Manager to kill the java process.

or,

2) You have some other app on your system which already uses the standard Creoson port (9056).  If so, you can change the Creoson port by either running the CreosonSetup GUI app, or by editing setvars.bat to change the JSON_PORT value.  If you do change this port, then any client code you have that calls Creoson will need to change to use the new port number.

yuyaU0507, 2024-03-11:
Thank you guys,

Yes, I've already run CreosonSetup.exe then got the error. I tried to start creoson_run.bat after I stopped the .exe, it works.
I also wondered how to stop the .bat actually. I'll try to kill java.exe process to do so.

closed 2024-03-11 by yuyaU0507