SimplifiedLogic/creoson #111: Usage of OTK C++ library in Version 2.9.0

tmp-user-dev, 2024-05-27:
The release note of Version 2.9.0 says that some of the function start using OTK C++ rather than J-Link.
https://github.com/SimplifiedLogic/creoson/releases/tag/v2.9.0

**Questions**
1. Does user need to install OTK C++ in addition to J-Link?
2. Does user need a OTK C++ license to unlock the application?

tmp-user-dev, 2024-05-29:
Thank you very much for the answer.

Correct me if my understanding is wrong.

**End Users** (who simply use distributed application which interact with creoson)
- Need to install JLink and creoson server
- No need to install OTK C++
- No license is required

**Developers** (who develops and distribute application which interact with creoson)
- Need to install JLink and creoson server
- Need to install OTK C++ ?
- Need OTK C++ license ?

If OTK C++ module is included in creoson and already unlocked plus the app connects to creo via creoson only, 
I'm bit confused why developers additionally need to install the C++ module and its license.

closed 2024-05-29 by tmp-user-dev

adama2000, 2024-05-29:
Sorry, what I said was misleading, I should have said it differently --

   - An OTK C++ license is needed if you are going to develop OTK C++
   libraries.

   - If you are developing your own app which uses the creoson library,
   then technically you are not an OTK C++ developer, you are merely using the
   OTK C++ library that is bundled with creoson.  So in that sense, your
   creoson developers are the same as end users.

   - As far as creoson is concerned, you would only need an OTK C++ license
   / installation if you were going to modify the C++ code that is inside
   creoson.  It's unlikely this is what you are planning to do, since most
   creoson developers just want to use the library as is.

Hope that clears things up,

Adam Andrews

adama2000, 2024-05-29:
That was probably too wordy.... to keep it simple: you don't need an OTK
C++ license to develop apps that use creoson, nor do you need to install
OTK C++.

Adam Andrews

tmp-user-dev, 2024-05-31:
Thanks for the clarification.
Now it's clear to me.