SimplifiedLogic/creoson #113: Creo 11.0.2.0 Read Parameter Erro

thackb, 2024-10-28:
Im testing Creo 11.0.2.0 with creoson 2.9 and getting the following error reading parameters from files:

xdr_ProParamvalue: Unknown entity type 0

[image]

adama2000, 2024-10-31:
I tested Creo 11.0.1.0 with a creoson "parameter list" command and I didn't
get that error.  Is that the creoson command that gives you the error, or
were you using a different function to get the error?

I can try downloading 11.0.2.0 later to try that version.  But I would like
more detail about what creoson functions you're calling.

Thanks,
Adam Andrews

thackb, 2024-10-31:
Hi Adam

I call the Parameter > List function with :-

{"sessionId":"-738066437541115818","command":"parameter","function":"list","data":{"file":"test.prt","name":"PART_NUMBER"}}

The JSON I get back from the function is :-

{"status":{"error":true,"message":"Cannot get parameter value using obsolete function; check that your PATH contains only one version of Creo"},"data":null}

thackb, 2024-10-31:
I have found the problem. My Fault :(
  
A Windows Enviorment Settings 

set PROE_COMMON=D:\PTC\Creo 11.0.2.0\Common Files

All working as normal now

closed 2024-10-31 by thackb