SimplifiedLogic/creoson #117: Filter bom_get_paths for ghost parts

jcp158, 2025-05-13:
I'm curious if it is possible to apply a filter to the get_paths or feature_list functions to sort out ghost parts. These are from references to parts that have been replaced with new components. They're still 'ACTIVE' status type and are still included in the feature list which I'm assuming is why they are picked up by both functions. They also have a feat#. I can find them manually using search, but haven't figured out a way to get this info programmatically. There is something that identifies them because they are not included in the asm model tree or included in the bill of materials command in Creo.

davidhbigelow, 2025-05-13:
Do you have a sample assembly that demonstrates this?

Dave

jcp158, 2025-05-13:
Sure. Do you mind sending me your email and I'll send you a link?

davidhbigelow, 2025-05-13:
***@***.***

jcp158, 2025-05-13:
Dave, Looks like it redacted your email. I think I found it elsewhere. Did you receive an email?

alex-levadnyy, 2025-05-22:
is this SimplRep?

jcp158, 2025-05-22:
> is this SimplRep?

No it's all Master Rep.