SimplifiedLogic/creoson #119: Request to Expand View Metadata: View Type, Parent-Child Relationships, Model View Reference for GEN_VIEW, Projection Direction for PROJ_VIEW, and Section View Attachments

babyhyf, 2025-06-24:
### **Issue Title**  
**Enhancement: Extend `drawing_list_view_details` to expose structured view metadata for AI-driven automation and historical data analysis**

---

### **Background**  
The current `drawing_list_view_details` method in `creopyson` returns view data **lacking critical metadata**, preventing accurate reconstruction of drawing view hierarchies.  
For example: inability to distinguish `GEN_VIEW` and `PROJ_VIEW`, missing projection directions (`left/right/top`), and unclear section view associations.  
This severely limits capabilities for **AI-powered drawing automation** and **historical drawing data mining**.

---

### **Problem Description**  
#### **Missing Fields**  
The current API response lacks the following key fields:  
```json
{
  "name": "DRAWING_FRONT",
  "type": "GEN_VIEW",          // View type (GEN_VIEW/PROJ_VIEW/SECTION)
  "parent": null,              // Parent view name (GEN_VIEW has no parent)
  "model_view": "TOP",         // Model standard view linked to GEN_VIEW (TOP/FRONT)
  "projection_direction": "right", // Projection direction for PROJ_VIEW (left/right/top/bottom)
  "section": [                // Section information (nullable)
    {"section name": "SECTION_A", "section_type": "Revolved"} // Section type (Full/Revolved/Half)
  ],
  "location": {"x": 100, "y": 100}, // Coordinates (unit: Drawing Units)
  "scale": "1:1"              // View scale
}
```

#### **Impact**  
1. **Ambiguous View Types**: Cannot distinguish between `GEN_VIEW` and `PROJ_VIEW`.  
2. **Missing Projection Directions**: Unable to determine projection relationships (e.g., "Right View" is a `PROJ_VIEW` with direction `right`).  
3. **Broken Section Information**: Cannot locate associated parent views or identify section types for section views.

---

### **Proposed Solution**  
#### **Goal**  
Enhance `drawing_list_view_details` or add a new `drawing_get_view_relations` method to return complete view hierarchy information.

#### **New Field Descriptions**  
| Field Name | Type | Description |
|------------|------|-------------|
| `type` | String | View type: `GEN_VIEW` (primary view), `PROJ_VIEW` (projected view), `SECTION` (section definition) |
| `parent` | String | Parent view name (only populated for `PROJ_VIEW`) |
| `model_view` | String | Model standard view linked to `GEN_VIEW` (e.g., `TOP`) |
| `projection_direction` | String | Projection direction for `PROJ_VIEW` (`left`, `right`, `top`, `bottom`) |
| `section` | Array | Section information (nullable):<br>- `section name`: Section name (e.g., `SECTION_A`)<br>- `section_type`: Section type (`Full`, `Revolved`, `Half`) |

#### **Example Output**  
```json
{
  "views": [
    {
      "name": "DRAWING_FRONT",
      "type": "GEN_VIEW",
      "parent": null,
      "model_view": "TOP",
      "section": [{"section name": "none", "section_type": "none"}],
      "location": {"x": 100, "y": 100},
      "scale": "1:1"
    },
    {
      "name": "PROJ1",
      "type": "PROJ_VIEW",
      "parent": "DRAWING_FRONT",
      "projection_direction": "right",
      "section": [{"section name": "none", "section_type": "none"}],
      "location": {"x": 150, "y": 100},
      "scale": "1:1"
    },
    {
      "name": "PROJ2",
      "type": "PROJ_VIEW",
      "parent": "DRAWING_FRONT",
      "projection_direction": "left",
      "section": [
        {"section name": "SECTION_B", "section_type": "Revolved"},
        {"section name": "SECTION_A", "section_type": "Revolved"}
      ],
      "location": {"x": 50, "y": 100},
      "scale": "1:1"
    }
  ]
}
```

---

### **Key Requirements**  
1. **Clear View Types**  
   - `GEN_VIEW`: Primary view (no parent).  
   - `PROJ_VIEW`: Projected view (requires `parent` and `projection_direction`).  
   - `SECTION`: Section definition (requires `section_type`).  

2. **Standardized Projection Directions**  
   - `projection_direction` must return directional values (`left`, `right`, `top`, `bottom`).  

3. **Clear Section Associations**  
   - Any view (`GEN_VIEW` or `PROJ_VIEW`) can link to sections via the `section` field.  
   - `section_type` must explicitly define the section type (`Full`, `Revolved`, `Half`, `Removed`).  

---

### **Use Cases**  
1. **Automated Drawing Layout Reconstruction**  
   - Rebuild standard view layouts using `model_view` and `projection_direction`.  
   - Example: `PROJ1` is the right view of `DRAWING_FRONT`.  

2. **Section Validation**  
   - Automatically verify that section views are correctly linked to parent views and annotated with types (e.g., `Revolved`).  

3. **Document Generation**  
   - Generate reports like `"PROJ1 is a left projection of FRONT with a revolved section SECTION_A"`.  

4. **AI Training Data Generation**  
   - **Scenario**: Extract structured view relationship data from historical drawings to train AI models for:  
     - **View Layout Prediction**: Input model geometry → Output optimal view arrangement rules.  
     - **Section Annotation Generation**: Automatically generate annotation symbols based on section types and parent view positions.  
     - **Drawing Intent Understanding**: Encode view hierarchies as feature vectors for semantic retrieval (e.g., "Find primary views with revolved sections").  

---

### **Implementation Notes**  
- **Technical Feasibility**: Creo internal data supports these metadata fields (accessible via J-Link). Suggest extending existing APIs.  
- **Compatibility**: Use `"none"` placeholder if certain fields cannot be retrieved (e.g., `section_name: "none"`).  

---

### **Why This Matters**  
These metadata fields are not only foundational for automated drawing processing but also **key enablers for unlocking the value of historical drawing data**.  
By using `creopyson` to batch-extract structured view relationship data, you can efficiently build large-scale training datasets for the following AI applications:

#### **1. Intelligent View Layout Generation**  
- **Problem**: Traditional CAD layout relies on engineer experience, which is time-consuming and error-prone.  
- **AI Solution**: Train models to predict optimal view arrangements using historical drawings’ view types (`GEN_VIEW/PROJ_VIEW`), projection directions (`projection_direction`), and coordinate relationships.  
- **Data Requirement**: Structured fields provide direct "input (model geometry) → output (view layout)" mappings.  

#### **2. Automated Section Annotation**  
- **Problem**: Manual annotation of section symbols and associations is inefficient.  
- **AI Solution**: Train models to recognize section cut paths and automatically generate annotations based on section information (`section.name/type`) and parent view positions.  
- **Data Advantage**: Fields explicitly distinguish section types (e.g., `Revolved/Half`), avoiding ambiguity in traditional image recognition.  

#### **3. Drawing Intent Understanding and Retrieval**  
- **Problem**: Searching for specific design patterns in massive drawing repositories is challenging.  
- **AI Solution**: Encode view relationships (e.g., `parent` hierarchy) as feature vectors for semantic intent-based retrieval (e.g., "Find drawings with revolved sections").  

#### **4. Automated Drawing Quality Inspection**  
- **Problem**: Manual consistency checks (e.g., view scales, projection directions) are costly.  
- **AI Solution**: Use structured data to train anomaly detection models that automatically identify conflicting rules (e.g., mismatched `projection_direction` and coordinate offsets in `PROJ_VIEW`).

davidhbigelow, 2025-06-24:
ChatGPT much?!

Would recommend contributing code that accomplishes the objectives for
review and acceptance to the library.  It is open source so… as long as the
functionality exists… and you follow the implementation… it should be
possible.

Dave

adama2000, 2025-07-06:
Unfortunately PTC's Creo API doesn't seem to support getting most of those values for drawing views.  Scale is possible, and location _might_ be -- the view's Transform is available.

babyhyf, 2025-07-21:
> Unfortunately PTC's Creo API doesn't seem to support getting most of those values for drawing views. Scale is possible, and location _might_ be -- the view's Transform is available.

Thank you for your feedback, i'm working on training a Ai to generate the 2D drawing,  it looks like that we have to wait or annotation by human