SimplifiedLogic/creoson #120: Can I run the server without launching the CreosonSetup.exe file?

jf-pds, 2025-06-26:
Is there a way I can run Creoson just by running a .bat without a terminal window? If I run creoson_run.bat when I close the window the java.exe process shuts down.

adama2000, 2025-07-06:
The only way I know of is to run the GUI application CreosonSetup.exe, press the "Start CREOSON" button, and then close the GUI window.  That won't open a terminal window, but the server will keep running in the background.

DieSwartKat, 2025-07-07:
I do this with a vbs script that along with a wscript file.

wscript.exe {path}silent.vbs {path}creoson_run.bat

_silent.vbs_
`
Set WinScriptHost = CreateObject("WScript.Shell")
WinScriptHost.Run Chr(34) & WScript.Arguments.Item(0) & Chr(34), 0
Set WinScriptHost = Nothing
`